profezzorn/ProffieOS #19: Lag when powering on saber, smooth swing fonts don't play correctly

SA-22C, 2019-07-01:
Using the current branch with all commits.  When igniting the saber, delays of 10-15 seconds are observed.  Once the saber is on, clash events work ok but smooth swing hum pairs don't play (just a medium pitched tone, similar to the sound when the Proffie hangs)  turning off the saber happens basically on time.  Switching presets works for a while but eventually the saber crashes and 'motion chip timeout, reboot motion chip' spams in the serial monitor.  Tried on three separate Proffie sabers, so it's unlikely to be a hardware issue. Reverting to 1.312 fixes the issue.

SA-22C, 2019-07-01:
Edit, upon further testing the errors are a bit inconsistent but the common thread is that there appears to be some issue with playback of wav files.  Sometimes it's the boot wav, sometimes the hum.  Sometimes it's the smooth swing pairs after a transition.  Eventually in all testing, the saber locks up and spams the motion chip timeout error.  I'm adding a short youtube video that captures some of the playback issues.

SA-22C, 2019-07-01:
https://youtu.be/TOC-eudBpaw

profezzorn, 2019-07-01:
Thanks for reporting this bug.
I haven't observed this bug myself, but I will try it and see if I can reproduce it.

If you have some time, you can help me narrow down when the problem occurred by using "git bisect".
(see https://git-scm.com/docs/git-bisect for more information.)
If you have trouble with git bisect, let me know.

SA-22C, 2019-07-02:
Used git bisect, looks like the commit for underflow fix is the culprit.

$ git bisect good
7ee6d7b7a6b5c70ea6777640b6e6c1888c53c144 is the first bad commit
commit 7ee6d7b7a6b5c70ea6777640b6e6c1888c53c144
Author: profezzorn <hubbe@hubbe.net>
Date:   Tue Jun 25 21:42:55 2019 -0700

    give new sound effects lower priority to avoid underflows

 sound/buffered_wav_player.h | 13 ++++++++++++-
 1 file changed, 12 insertions(+), 1 deletion(-)

profezzorn, 2019-07-02:
Thank you, That is very helpful!
I guess I'm not surprised that this change is the one that causes problems, but I'm not sure I get why.
It occurs to me that my arduino-proffieboard plugin has some changes that may be part of this problem too. Hopefully I'll have time to do some investigating tonight.

profezzorn, 2019-07-03:
I think the motion chip timeout is a problem might be caused by setting the processor clock very low without some of the fixes that I made in the arduino-proffieboard plugin. I added a potential workaround for this in https://github.com/profezzorn/ProffieOS/commit/d1e570762c9a3f4f48a2b68d909fb809e94b7f6c If you have time to try it, please let me know if it helps.

SA-22C, 2019-07-03:
Ok, did a very quick test and so far the issue seems resolved.  Doing some more testing later today but so far so good.

profezzorn, 2019-07-03:
Great!
Although there is a possibility that the problem comes back with the next arduino-proffiboard release.
I'll see if I can put up a beta of the arduino-proffieboard plugin this weekend, and maybe you can try it out and see if that works well too.

SA-22C, 2019-07-04:
I spoke too soon. My earlier tests were of non smoothswing fonts which work fine. As soon as I tested KSith’s Light font, the delays came back with a vengeance.

profezzorn, 2019-07-04:
That could explain why I haven't seen any problems. I've mostly been working with thermal detonator fonts recently.

SA-22C, 2019-07-04:
Yeah, the underflow commit doesn’t play well with smoothswing.

profezzorn, 2019-07-04:
I may have figured out what was wrong with the underflow commit.
Potential fix here: https://github.com/profezzorn/ProffieOS/commit/f9c71da4b036374e073bf0605a967e1a96892ff3

Please let me know if it works for you.

SA-22C, 2019-07-04:
Tested three fonts: NEC, Plecter and Smoothswing.  All three seem good.  I'll so some more testing with accent swings later today.

profezzorn, 2019-07-05:
Well the underflow optimizations have certainly been a lightning rod for audio bugs.
I just found and fixed another one.

Anyways, hopefully this bug is fixed now, so I'm going to close it. Let me know if you see more problems.

closed 2019-07-05 by profezzorn