profezzorn/ProffieOS #20: Audio glitches on 2.0 BETA

SA-22C, 2019-07-10:
Similar audio issues to #19 but not sure if it's a new issue or the old one resurfacing.

Symptoms:
- when using a smooth swing font (ie: Light by KSITH) the ignition will sometimes be delayed up to 10 seconds
- hums will sometimes not play, or hang with a high pitched tone
- blasters do not play when the button is pressed rapidly
- accent swings lag and freeze
- poweroffs don't play and are replaced with a loud medium-pitched tone.

A video is uploading now and I'll start doing a git bisect.

profezzorn, 2019-07-10:
If I was a betting man (which I am not), I would suspect that these problems stem from:

https://github.com/profezzorn/ProffieOS/commit/fc0ab5ffc74d2cd995b6234d44e694ef3887fe71 and 
https://github.com/profezzorn/ProffieOS/commit/e02043e03063ab8ab3e6b07a1f13a4a42f3371fc

(Those two were really meant to be one commit.)

They are meant to fix another problem where wav players can have Stop() called from an interrupt which can cause a lot of problems.

SA-22C, 2019-07-10:
That's a good place to start.  Thanks.

https://youtu.be/w8j_dFemSgQ

SA-22C, 2019-07-10:
I'm getting glitches all the way back to the commit flagged in my original issue.  Using code after 918aeab produces the bug.  I'll try to use bisect to figure out the cause but so far I've been unsuccessful.

SA-22C, 2019-07-10:
$ git bisect bad
7ee6d7b7a6b5c70ea6777640b6e6c1888c53c144 is the first bad commit
commit 7ee6d7b7a6b5c70ea6777640b6e6c1888c53c144
Author: profezzorn <hubbe@hubbe.net>
Date:   Tue Jun 25 21:42:55 2019 -0700

    give new sound effects lower priority to avoid underflows

 sound/buffered_wav_player.h | 13 ++++++++++++-
 1 file changed, 12 insertions(+), 1 deletion(-)

Matthew@DESKTOP-PGUMJA9 MINGW64 ~/Desktop/proffieosbugfix/ProffieOS ((7ee6d7b...)|BISECTING)

Back to the underflow commit.

profezzorn, 2019-07-10:
Makes sense unfortunately.
That CL might not exactly cause the bugs, but it sure makes them worse...

profezzorn, 2019-07-10:
Does it seem to happen more on some fonts?
Does it happen with the stock fonts? (That's probably what I'll be using when I try to reproduce this problem myself tonight...)

SA-22C, 2019-07-10:
I just tested Dark Kyber and Smooth Fuzz.  Both exhibited the same bugs.  Easiest way to trigger the bugs is to hammer on the aux button to try and trigger rapid blaster shots.  This will cause the motion chip timeout/reboot issue and glitch out the audio.

profezzorn, 2019-07-11:
Good news/bad news: It's acting really wonky for me too.

profezzorn, 2019-07-11:
I did a manual bisect and got the same result as you.
Now, if I could just figure out *why*....

profezzorn, 2019-07-11:
Fixed in https://github.com/profezzorn/ProffieOS/commit/9cb2b6c07c87668dcbe326a31a532393bd1e8161 I think. Please try it out and let me know. I will be releasing a new BETA shortly.

closed 2019-07-11 by profezzorn