profezzorn/ProffieOS #21: Hum gets stuck playing on 0 Buttons mode

DonSaber, 2019-07-11:
I verified that the problem with the hum sound getting stuck playing is still present on release 2.1.
This time I tried to investigate a bit deeper, using the serial monitor and I believe I got some clues.

Below you can find the logs taken during a sequence of Twist ON / Twist OFF / Twist ON.

TWIST
EVENT: Twist millis=40935
Ignition.
unit = 0 vol = 0.00, Playing SmthJedi/hum01.wav
channels: 1 rate: 44100 bits: 16
unit = 1 vol = 0.50, Playing SmthJedi/out01.wav
channels: 1 rate: 44100 bits: 16
unit = 2 vol = 0.00, Playing SmthJedi/swingl06.wav
channels: 1 rate: 44100 bits: 16
unit = 3 vol = 0.00, Playing SmthJedi/swingh06.wav
channels: 1 rate: 44100 bits: 16
unit = 2 vol = 0.00, Playing SmthJedi/swingl05.wav
channels: 1 rate: 44100 bits: 16
unit = 3 vol = 0.01, Playing SmthJedi/swingh05.wav
channels: 1 rate: 44100 bits: 16
TWIST
EVENT: Twist ON millis=42661
unit = 1 vol = 0.50, Playing SmthJedi/in01.wav
channels: 1 rate: 44100 bits: 16
TWIST
EVENT: Twist millis=42777
Ignition.
unit = 4 vol = 0.00, Playing SmthJedi/hum01.wav
channels: 1 rate: 44100 bits: 16
unit = 5 vol = 0.50, Playing SmthJedi/out01.wav
channels: 1 rate: 44100 bits: 16
Playing SmthJedi/hum01.wav
channels: 1 rate: 44100 bits: 16
Playing SmthJedi/hum01.wav
channels: 1 rate: 44100 bits: 16
unit = 2 vol = 0.00, Playing SmthJedi/swingl02.wav
channels: 1 rate: 44100 bits: 16
unit = 3 vol = 0.00, Playing SmthJedi/swingh02.wav
channels: 1 rate: 44100 bits: 16

Apparently, the third event (Twist ON) was detected before the board actually completed the shutdown process.
Basically, when the hum gets stuck I don't see the two following lines, which are normally printed in a "normal" Twist OFF process:

Amplifier off.
Unmounting SD Card.

To reproduce the issue I found very useful to keep the saber in vertical position holding it from the blade. Holding the saber from the hilt makes it very difficult to reproduce the problem, probably because it is harder to repeat twists in rapid sequence.
Best way is to hold the blade between your hands and rub them, causing the saber to twist.

profezzorn, 2019-07-11:
Note that "TWIST" and "Twist ON" are not separate events, but two printouts from the same event.
And "Twist On" just means that a twist event was received while the saber was ON.
So the "Twist ON" is only the second event listed above, and it is the event that triggers the shutdown.
From what it looks like, there is another twist event 116 ms after the "Twist ON" event, which turns the saber on again. So it seems like there is something wrong with the twist detection code, because it shouldn't be possible to get another twist event without performing the left-right-left (or right-left-right) motion again.

I'm going to see if my zero-button teensysaber toy saber is still working and see if I can give this a try.

DonSaber, 2019-07-11:
Many thanks, I typed Twist On / Off to give you a reference about the saber state while performing the test. In my experience, the hum gets 'stuck' only when the saber is turning off.
Again, best way to try and reproduce the problem is to hold the saber vertically with the blade in your flat hands and rub them. This is causing some fast twisting, that very likely are detected before the shutdown process is completed.
I was never able to reproduce the problem on purpose twisting the saber in orizontal position, probably because I cannot twist it fast enough this way.
However, the bug occurred several times while using the saber for sparring, but in these cases I'm not able to identify the motion that triggers the bug because I'm too concentrated on the sparring.
If you need any other info, just let me know.

profezzorn, 2019-07-11:
It's unfortunate that this occurs during sparring.
That means that the twist detector is detecting things that weren't meant to be twists.
Maybe I can make it bettery by requiring that the swing speed be very low during twisting.

DonSaber, 2019-07-11:
Well, actually we have tecniques or combinations of movements that for many reasons include twists, so probably the saber just reacted to some of these movements in certain conditions.
Perhaps it is a good idea to limit the activation/deactivation to slow twists, that likely are less probable to occur while sparring.

BoykoKazakov, 2020-01-02:
Maybe you can use the (https://www.youtube.com/watch?v=vWXiVbZWoTE) polaris saber motion/posture idea. After all those guys are huge fighting club and have some close experience with the no button sabers

DonSaber, 2020-01-04:
Many thanks for your suggestion.
In the end I disabled the 0-Buttons feature and returned to a more classic 1-Button, far more realiable in my scenario.
By the way, I'm part of Ludosport since 9 years and I know very well how a Polaris works.
The gesture activation system has its pro and cons and it is widely appreciated by the pupils of the Academy, but I remain a strong supporter of the push buttons.
As last info, I'm still getting random sound lock-ups using the SA-22C version.
Not yet tested version 3.6, but will try it in the near future.

profezzorn, 2020-05-18:
I this still an issue?  (sorry for the long delay here...)

DonSaber, 2020-05-19:
I just tried with ver. 4.4 Beta and I can confirm that I was not able to replicate the issue anymore.
Looks like the problem is solved.

closed 2020-05-19 by profezzorn