profezzorn/ProffieOS #66: Why does the OS prefer font sounds in subfolder structure?

NoSloppy, 2020-07-07:
Just adding this issue as a reminder that for some reason, there are more Audio Underflows and blade performance hits when all the wavs are in one folder together as opposed to grouped into subfolders of like sounds.

profezzorn, 2020-07-07:
Read up on the FAT32 format.
It seems like each block can contain 16 file entries.
So, if you have 128 files in one directory, it has to read on average 4 blocks to find the the file.
If you have 16 subdirectories with 16 files each, only 2 blocks has to be read.
It may be possible to optimize this by adding most commonly used files(hum) first.
Also, ProffieOS might be able to improve on this with some caching.

For effects that only have one sound, it should be more efficient to not put them into a sub-directory.

NoSloppy, 2020-08-18:
Seems like this can be closed for now?

profezzorn, 2020-08-18:
I'm still contemplating ways to make file lookups faster. :)

closed 2022-12-29 by profezzorn