profezzorn/ProffieOS #68: bugs in style_parser.h

artandtechvof, 2020-08-12:
Using the WebUSB app I noticed some bugs in the behavior of the selectable built-in presets
The named styles are;

- standard -> functions as expected, and reports correct parameters when asked to  describe_named_style
- advanced ->  functions as expected, and reports correct parameters when asked to  describe_named_style
- fire ->  functions as expected, and reports correct parameters when asked to  describe_named_style
- unstable -> has duplicate parameter for extension and retraction 
                 ....InOutFuncX<IntArg<5, 100>,IntArg<5, 200>>,Black>>(), and therefore the describe_named_style is also incorrect (missing 1 parameter)
  correct;  
....InOutFuncX<IntArg<5, 100>,IntArg<6, 200>>,Black>>()

- strobe -> has bug, enumeration of parameters is incorrect and has duplicate;
       original:  StyleNormalPtrX<StrobeX<RgbArg<1, BLACK>, RgbArg<2, WHITE>, IntArg<3, 15>, IntArg<5, 1>>, Rainbow, IntArg<1, 300>, IntArg<2, 800>>(),
      ( extend and retract parameter are coupled to standby and flash color (resulting in extreme slow retraction...), Arg<4 is skipped all together )
      correct: StyleNormalPtrX<StrobeX<RgbArg<1, BLACK>, RgbArg<2, WHITE>, IntArg<3, 15>, IntArg<4, 1>>, Rainbow, IntArg<5, 300>, IntArg<6, 800>>()

- cycle: although the style is working, the describe_named_style is somewhat cryptic;
  "Cycle blade, start color, base color, flicker color, blast color, lockup color" (5 parameters according to describe_named_style)
  In practice base color and flicker color are the same parameter, better description would be;
  "Cycle blade, start color, base color/flicker color, blast color, lockup color"
  Further more, this style never fully comes to a solid blade, because the max percentage of the blade is set to 90  
  Resulting in a flickering blade on full extension. Also the base color is an Audioflicker of the start and base color, was this 
  intentional?
 NOTE: cycle would be nice to also include the cycle speed, and transition time. This is and easy fix for my own use, but would be 
 nice to see in the main core code.
  
- rainbow ->  functions as expected, and reports correct parameters when asked to  describe_named_style
- charging ->  functions as expected, and reports correct parameters when asked to  describe_named_style
- builtin ->  functions as expected, and reports correct parameters when asked to  describe_named_style

All tested on Teensy and Proffie V2 with serial commands and WebUSB app.    

ObiRobKenobi,  Art and Tech VoF.

profezzorn, 2020-08-14:
I had a go at fixing these, but I haven't had time to test them properly.
If you get a change to try it, let me know.

artandtechvof, 2020-08-14:
No problem. I am testing a lot with the WebUSB anyway, so testing the bugfixes is no problem.
I will report any issues, or that everything is working asap.

artandtechvof, 2020-08-14:
Tested all the options/styles via WebUSB. All styles are now working as expected. Thanks for the quick fix!

If possible, it would be nice to have the 'cycle'-style up and down time for the 'cycle' preset also configurable. It is basically the extension and retraction style of this preset, and is a quick fix to add to the configurable parameters;

 { "cycle",
    StylePtr<ColorCycle<RgbArg<1, Blue>,0,1,Layers<
        AudioFlicker<RgbArg<3, Cyan>, RgbArg<2, Blue>>,
        BlastL<RgbArg<4, Rgb<255,50,50>>>,
        LockupL<HumpFlicker<RgbArg<5, Red>, RgbArg<3, Cyan>,100>>,
        SimpleClashL<White>>,
      100,IntArg<6, 2000>,IntArg<7, 1000> >>(),
    "Cycle blade, start color, base color, flicker color, blast color, lockup color, cycle up time, cycle down time"
  },

Just a nice to have. If you don't feel like making this adjustment, this issue can be closed.

profezzorn, 2020-08-14:
Unfortunately, it's not quite that simple, because ColorCycle<> takes simple integers as inputs, not classes. I would need to make an ColorCycleX<> that takes the right arguments first. It's not terribly difficult, but not quite as easy as you suggest.

closed 2020-08-14 by profezzorn