profezzorn/ProffieOS #82: Request: Mount/unmount SD manually

NoSloppy, 2020-08-28:
Is it possible to disable auto-mount on connection and just have a serial monitor command to mount the SD card on demand if USB + Mass Storage is programmed?
Maybe a toggle for those who want to leave auto-mount on? I personally go insane waiting for the card to mount only to unmount it immediately many times during testing, and another option besides completely killing the mount option in Arduino could be cool.

profezzorn, 2020-08-28:
Maybe...
Technically, the computer initiates the mount, but ProffieOS can deny it, which is exactly what happens if ProffieOS is using the SD card.

Personally I keep my sabers programmed for just "Serial" most of the time, and turn on mass storage when I need it.
Of course, I also use linux, so I don't really have this problem, since it doesn't auto-mount USB drives unless you run some program that does that for you.

profezzorn, 2020-09-04:
I wonder if for your purposes it would be enough to just change this line:
https://github.com/profezzorn/ProffieOS/blob/5f8bfe4a808c7a1fa356e45b0a836840c54f5bc0/common/sd_card.h#L26

It has a 1-second timeout before it releases the SD card so that the computer can mount it.
Maybe if you set it to 60 seconds or something, you have time to upload or whatever it is you want to do before the computer has an opportunity for mounting the card?

NoSloppy, 2020-09-05:
Thank you. I'll probably just do as you first suggested and turn off Mass Storage until l need it. But this is good to have up my sleeve if I'd want to try it out.

closed 2024-07-07 by profezzorn