profezzorn/ProffieOS #89: x_config.h : when maxLedsPerStrip # matches WS281XBladePtr<#,> blade does not light up & no errors

artandtechvof, 2020-09-07:
Weird phenomenon and also not sure if its intentional or a bug, but when the number of leds defined in the config file;

const unsigned int maxLedsPerStrip = #nr;

matches exactly with the number in the BladeConfig

 WS281XBladePtr<#nr,,,>

Compilation does not throw an error, code is compiled and uploaded, but the blade is not ignited, sound is playing and everything else seems to work. No weird warnings in serial monitor.

When the maxLedsPerStrip # < WS281XBladePtr<> 
Serial monitor however repeatedly complaining about the Display memory not being large enough.

Only when maxLedsPerStrip nr#+1 > WS281XBladePtr<#nr,,,> Everything is working.

(I seem to remember that for older versions of TeensySaber/ProffieOS there was a compile time error to prevent upload when not enough leds were define?)

profezzorn, 2020-09-08:
This should be fixed now. (Haven't had time to test it properly though...)

closed 2020-09-08 by profezzorn

artandtechvof, 2020-09-08:
Works perfect! When the maxLedsPerStrip # is less than WS281XBladePtr<> a compile time error it shown in Arduino, and compilation stops.
With an equal number compilation is error free, and saber works as expected.

Thx for fixing this so quickly!