profezzorn/ProffieOS #99: Twist Off... Audio humming intermittently remains on.

justbill2020, 2020-09-23:
There are times with ProffieOS v5 when I use the twist-off gesture that the blade goes off however the humming noise from the audio remains on. I might need some guidance as I'm new to this code base but I can help debug and contribute to solve this issue.

Is there a way to log data to a text file on the SD card with a time stamp to debug what's happening? 

Or can we expand the serial mon to work with Bluetooth?

profezzorn, 2020-09-23:
The serial monitor already works with bluetooth.
There is a command called "make_default_consol" which makes that serial port get all the asynchronous messages that ProffieOS generates. Then you just need the other side of the bluetooth connection to work like a serial port, and select that serial port in Arduino->Tools->Port (or some other serial communication program)

There is also a handy "whatison" command which can tell you a little bit about what wav player(s) are on and what they are playing.

profezzorn, 2020-10-03:
I've made some changes to twist recently that *may* resolve this problem.
Does it still occur for you?

profezzorn, 2021-02-25:
I'm closing this issue since there was no reply to my question.

closed 2021-02-25 by profezzorn

justbill2020, 2021-02-25:
yea i'm so sorry i've been so busy I haven't been able to test, and you've
made so many updates i'm sure its now resolved.