profezzorn/ProffieOS #101: Stripes compiling error with latest commits

NoSloppy, 2020-10-02:
Hi.
I was sequentially adding the latest commits and verifying after each one using my config.
When I got up to cf9e561 initial nod implementation, it no longer compiled. I tried the default config and it worked. 
I then changed one of the default styles to 
StylePtr<Stripes<1000,1000,Cyan,Magenta,Yellow,Blue>>(), 
it failed then and this is the error generated:

`In file included from /Volumes/Media Master/Dropbox/Lightsabers/Soundboards/Proffieboard/a_Unofficial releases/ProffieOS_5.x/ProffieOS/ProffieOS.ino:392:0:
/var/folders/s2/14yvr53n76xbvx2ctcxwhrvc0000gn/T/arduino_build_477175/sketch/styles/stripes.h: In instantiation of 'void StripesHelper<A, B ...>::get(int, int, SimpleColor*) [with A = Rgb<0, 255, 255>; B = {Rgb<255, 0, 255>, Rgb<255, 255, 0>, Rgb<0, 0, 255>}]':
/var/folders/s2/14yvr53n76xbvx2ctcxwhrvc0000gn/T/arduino_build_477175/sketch/styles/stripes.h:79:5:   required from 'SimpleColor StripesBase<COLORS>::getColor(int) [with COLORS = {Rgb<0, 255, 255>, Rgb<255, 0, 255>, Rgb<255, 255, 0>, Rgb<0, 0, 255>}]'
/var/folders/s2/14yvr53n76xbvx2ctcxwhrvc0000gn/T/arduino_build_477175/sketch/styles/style_ptr.h:41:42:   required from 'void Style<T>::run(BladeBase*) [with T = StripesX<Int<1000>, Int<1000>, Rgb<0, 255, 255>, Rgb<255, 0, 255>, Rgb<255, 255, 0>, Rgb<0, 0, 255> >]'
/Volumes/Media Master/Dropbox/Lightsabers/Soundboards/Proffieboard/a_Unofficial releases/ProffieOS_5.x/ProffieOS/ProffieOS.ino:1796:1:   required from here
/var/folders/s2/14yvr53n76xbvx2ctcxwhrvc0000gn/T/arduino_build_477175/sketch/styles/stripes.h:42:40: error: conversion from 'OverDriveColor' to non-scalar type 'SimpleColor' requested
       SimpleColor tmp = a_.getColor(led);
                                        ^`

NoSloppy, 2020-10-02:
USER ERROR - IGNORE, SORRY

NoSloppy, 2020-10-02:
Edit. Narrowing down further, I commented out all but one style in my config, and added back in one layer at a time. It works until I add a stab transition effect containing an AlphaL. Once that is uncommented, this error happens:
`ketch/styles/transition_effect.h:51:8: error: no matching function for call to 'TrConcat<TrWipeInX<Int<300> >, AlphaL<Rgb<128, 0, 0>, SmoothStep<Int<20000>, Int<2000> > >, TrJoin<TrSmoothFadeX<Int<900> >, TrWipeX<Int<700> > > >::getColor(RGBA_um_nod, RGBA_um_nod, int)'
   auto getColor(int led) -> decltype(transition_.getColor(RGBA_um_nod::Transparent(),
        ^`

profezzorn, 2020-10-02:
Please provide an actual example of a style that doesn't work.

NoSloppy, 2020-10-02:
Sure, sorry about that.
StylePtr<Layers<
Green,
TransitionEffectL<TrConcat<TrWipeIn<300>,AlphaL<Rgb<128,0,0>,SmoothStep<Int<20000>,Int<2000>>>,TrJoin<TrSmoothFade<900>,TrWipe<700>>>,EFFECT_STAB>,
    InOutTrL<TrWipeSparkTip<White,250>,TrWipeInX<Int<700>>>>>(),

profezzorn, 2020-10-02:
Fixed: https://github.com/profezzorn/ProffieOS/commit/35f58a694e556fc94356e3c98481ff1eabe29635

closed 2020-10-02 by profezzorn

NoSloppy, 2020-10-02:
Cool, thanks.
Now the error changed to this:
`In file included from sketch/styles/sparkle.h:16:0,
                 from /Volumes/Media Master/Dropbox/Lightsabers/Soundboards/Proffieboard/a_Unofficial releases/ProffieOS_5.x/ProffieOS/ProffieOS.ino:365:
sketch/styles/alpha.h: In instantiation of 'decltype ((((AlphaL<COLOR, ALPHA>*)this)->AlphaL<COLOR, ALPHA>::color_.getColor(led) * ((AlphaL<COLOR, ALPHA>*)this)->AlphaL<COLOR, ALPHA>::alpha_.getInteger(led))) AlphaL<COLOR, ALPHA>::getColor(int) [with COLOR = Rgb<0, 0, 0>; ALPHA = InOutHelperF<InOutFuncX<Int<300>, Int<800> >, true>; decltype ((((AlphaL<COLOR, ALPHA>*)this)->AlphaL<COLOR, ALPHA>::color_.getColor(led) * ((AlphaL<COLOR, ALPHA>*)this)->AlphaL<COLOR, ALPHA>::alpha_.getInteger(led))) = RGBA_um_nod]':
sketch/styles/layers.h:38:32:   required from 'decltype ((((Layers<BASE, L1>*)this)->Layers<BASE, L1>::base_.getColor(led) << ((Layers<BASE, L1>*)this)->Layers<BASE, L1>::layer_.getColor(led))) Layers<BASE, L1>::getColor(int) [with BASE = Layers<Rgb<0, 0, 0>, SimpleClashL<Rgb<0, 0, 0>, 40, (BladeEffectType)1, SmoothStep<Int<16384>, Int<24000> > >, LockupL<AlphaL<Rgb<255, 255, 255>, NoisySoundLevelCompat>, AlphaL<Rgb<255, 255, 255>, NoisySoundLevelCompat>, Int<32768>, Int<32768>, Int<32768> >, AlphaL<Rgb<255, 255, 255>, BlastF<200, 100, 400, (BladeEffectType)2> > >; L1 = AlphaL<Rgb<0, 0, 0>, InOutHelperF<InOutFuncX<Int<300>, Int<800> >, true> >; decltype ((((Layers<BASE, L1>*)this)->Layers<BASE, L1>::base_.getColor(led) << ((Layers<BASE, L1>*)this)->Layers<BASE, L1>::layer_.getColor(led))) = OverDriveColor]'
sketch/styles/style_ptr.h:47:42:   required from 'void Style<T>::run(BladeBase*) [with T = Layers<Layers<Rgb<0, 0, 0>, SimpleClashL<Rgb<0, 0, 0>, 40, (BladeEffectType)1, SmoothStep<Int<16384>, Int<24000> > >, LockupL<AlphaL<Rgb<255, 255, 255>, NoisySoundLevelCompat>, AlphaL<Rgb<255, 255, 255>, NoisySoundLevelCompat>, Int<32768>, Int<32768>, Int<32768> >, AlphaL<Rgb<255, 255, 255>, BlastF<200, 100, 400, (BladeEffectType)2> > >, AlphaL<Rgb<0, 0, 0>, InOutHelperF<InOutFuncX<Int<300>, Int<800> >, true> > >]'
/Volumes/Media Master/Dropbox/Lightsabers/Soundboards/Proffieboard/a_Unofficial releases/ProffieOS_5.x/ProffieOS/ProffieOS.ino:1795:1:   required from here
sketch/styles/alpha.h:36:49: error: could not convert 'RGBA_um::Transparent()' from 'RGBA_um' to 'RGBA_um_nod'
     if (alpha == 0) return RGBA_um::Transparent();
                                                 ^`

profezzorn, 2020-10-02:
Try a git pull, alpha.h has not been updated correctly.

NoSloppy, 2020-10-02:
Confirmed all working. Plus I'm seeing the space savings, thanks!