profezzorn/ProffieOS #103: Request for future: less repeated random

NoSloppy, 2020-10-04:
How could we make a note of the force.wav playing, so that the next time force is triggered, it does not play that exact one again?

profezzorn, 2021-02-24:
This is now implemented, activated with #define NO_REPEAT_RANDOM

closed 2021-02-24 by profezzorn

NoSloppy, 2021-02-24:
Oh that's great, and for ALL sounds too, not just force....of course. Yay,
thanks so much!

NoSloppy, 2021-03-16:
Sorry to dig this up. I just had repeats of push.wav for force push. There are only 2 push sounds.
 Is the no-repeat-random somewhat defeated if there are only 2 files to choose from?

profezzorn, 2021-03-16:
I figured it would be boring if it would just flip back and forth all the time if you only have two files, so I made it so that if you have 2 files, there is a 25% chance of a repeat instead of a 50% chance of a repeat.