profezzorn/ProffieOS #117: Future request: OLED screen 180 flip

NoSloppy, 2020-10-31:
I'm terribly sorry I don't know how to participate in pull requests yet, but I found that this simple edit would allow for OLED displays to flip 180 degrees. Tested and working fine by multiple people along with myself.
Change to following lines in ssd1306.h:

Send(SEGREMAP | 0x1);
Send(COMSCANDEC);
change to:
#ifndef OLED_FLIP_180
Send(SEGREMAP | 0x1);
Send(COMSCANDEC);
#else
Send(SEGREMAP);
Send(COMSCANINC);
#endif
Then add #define OLED_FLIP_180 to your config file and comment/uncomment it as needed.

Aleyan82, 2020-10-31:
I tested the feature and it makes a very nice addition to the OLED settings!!!

Aleyan82, 2020-11-01:
I made a pull request for this option in the main repo.

closed 2020-11-02 by NoSloppy