profezzorn/ProffieOS #126: WebBLE USB freezes on current ProffieOS master version

artandtechvof, 2021-01-21:
When trying to connect via WebUSB or WebBLE, the current github master of ProffieOS freezes on connecting.

It hangs on the describe_named_style builtin, this does return 
builtin preset styles, preset num, style index
INT 0
INT 1

But after that the board becomes unresponsive (also for buttons or any serial output)

The styles/style_parser has a couple of new line added compared to OS5.7 release;

line 35:  ArgParser ap(SkipWord(CurrentArgParser->GetArg(2, "", "")));
line 34:  CurrentArgParser = &ap;

When these lines are commented out, everything works as expected, and no adverse effect on the functioning of ProffieOS was noticeable.

Not sure to make this a pull request, as I do not know why the lines are there, and it is very easy to correct if these lines can be deleted?

profezzorn, 2021-01-21:
I actually changed those lines yesterday for unrelated reasons.
Please do a git pull try again and see if the problem still persists.

artandtechvof, 2021-01-21:
Ye I just saw it after posting this issue.

Fresh git pull, everything working as it should after upload, closing this issue.

closed 2021-01-21 by artandtechvof