profezzorn/ProffieOS #147: Request: Use a variable in blade config for tracks that are within the SD font folder

FrankKerschbaumer3, 2021-03-28:
With font makers having tracks that are usually contained within the font folder, would it be possible to reference the path of the font folder in the tracks path without having to reference the full path?

Example:

`{ "LS6", "LS6/tracks/jabba_scene.wav"`
to
`{ "LS6", "$PWD/jabba_scene.wav"`

Or is it recommended to have all of the tracks in the root dir of the SD card?

Thank you!

profezzorn, 2021-03-28:
Why? It's not hard to just enter the full path.

FrankKerschbaumer3, 2021-03-28:
Not hard, just tedious when dealing with several blade styles. Hence why I just wanted to ask 👍

profezzorn, 2021-03-28:
There is some perceived value, but I don't think it's worth the extra code.

closed 2021-03-28 by profezzorn