profezzorn/ProffieOS #199: OLED Stuck on message.

NoSloppy, 2021-05-30:
Switching presets before the ProffieOSFontImageDuration expires works fine.
If allowed to show for duration, it remains on the screen and is stuck there until power cycle.

NoSloppy, 2021-05-30:
Watching serial monitor, it looks like maybe I2C is sleeping before or too closely to the display duration time?

NoSloppy, 2021-05-30:
Testing more, it returns to battery meter *sometimes* but seems more likely to get stuck after sitting for a bit.
Maybe a video would be good:

https://photos.app.goo.gl/tSj44mDkBwmBCjSD8

NoSloppy, 2021-09-14:
Confirmed still doing this.

profezzorn, 2021-09-16:
I sat down to try to reproduce this problem, but I didn't even get that far before I found a bunch of bugs.
Anyways, I don't think the stuff that I changed fixes your problem, but if you get a chance to re-test, please let me know how it went.

NoSloppy, 2021-09-16:
FYI it didn't like STDERR undeclared in this scope. Commented out the remaining ones and compiled ok.
Unfortunately, it did not fix the freezing message. Can I help sniff out the issue?  
Maybe something about ProffieOSFontDuration? Because I also have an issue with that when using the idle.bmp in lieu of battery meter, although it's the opposite behavior; it only blips the text then shows idle instead of playing font for the set Duration.

profezzorn, 2021-09-17:
What I really need is more information.
Like, maybe post your font config file.
Is your image animated? (Does it matter if it is or not?)
(Maybe post the animation you're using.)
Do you have any modifications in your ProffieOS code? (Does it do the same thing without those modifications.)
Also, do you have idle times and stuff like that configured in your config file?
How the motion chip is being used might matter if this is related to i2c sleeping.

NoSloppy, 2021-09-17:
> How the motion chip is being used might matter if this is related to i2c sleeping.
Ok It happened but gdb says no stack so I must have done something wrong with the order of that boot-the-board-start-gdb process.
However, when the message froze, serial monitor **does indeed** mention I2C sleeping when it happened.
```
:12:55.473 -> DISPLAY: length
23:12:55.473 -> finder
23:12:55.473 -> Scanning sound font: common done
............................
23:12:55.825 -> unit = 0 vol = 0.50, Playing common/font01.wav
23:12:55.825 -> channels: 1 rate: 44100 bits: 16
23:12:55.825 -> LEN=1
23:12:57.754 -> Amplifier off.
23:12:57.862 -> I2C sleeping..
23:12:58.745 -> Unmounting SD Card.
23:13:02.849 -> I2C init..
23:13:04.823 -> I2C sleeping..
```
> Like, maybe post your font config file.
https://www.dropbox.com/s/a950o5a3sx3v3d9/config.ini?dl=0

> Is your image animated? (Does it matter if it is or not?)
The issue is that the text message gets stuck. There is no font image in the font.

> Do you have any modifications in your ProffieOS code? (Does it do the same thing without those modifications.)
Of course I do LOL. However, these test are all on current Github master clone with just my Testbed config copied over.

> Also, do you have idle times and stuff like that configured in your config file?
I have motion and IDLE_OFF times set to 2 minutes, but this frozen message happens before that.

profezzorn, 2021-09-19:
I think this should be fixed now. Please try it and update this issue as needed.

NoSloppy, 2021-09-19:
> I think this should be fixed now. Please try it and update this issue as needed.

Seems like freezing is fixed. The battery meter doesn't stay on though. Is that intentional?
https://www.youtube.com/watch?v=3IsIfPk7_OE

profezzorn, 2021-09-19:
That was not intentional. Should be fixed now.

NoSloppy, 2021-09-20:
Battery meter stops displaying with this code in `case SCREEN_PLI:` (as expected), and things work fine with it commented out....
```
	if (!SaberBase::IsOn() && t_ > font_config.ProffieOSFontImageDuration) {
	  screen_ = SCREEN_OFF;
	  return FillFrameBuffer2(advance);
	}
```

profezzorn, 2021-09-20:
Without that code, the battery meter would be on all the time, even when the saber is not ignited. That doesn't seem right.

To be honest, I don't remember how it used to work though....

NoSloppy, 2021-09-20:
> Without that code, the battery meter would be on all the time, even when the saber is not ignited. That doesn't seem right.
> 
> To be honest, I don't remember how it used to work though....

That is correct, and that is how it worked in the past. Battery meter shows all the time when blade is off, before IDLE_OFF_TIME. (Unless idle.bmp comes into play. My other PR is about to be closed and new improved submitted.

profezzorn, 2021-11-03:
Is this still an issue?

NoSloppy, 2021-11-03:
Well the original issue was the message was stuck when i2c went to sleep. That seems fine as of Sept 19.
The 2 issues right now are 1.an actual issue and 2. a change from how it used to work that you introduced.
#2 is mentioned recently (above) where we 're now doing SCREEN_OFF when t > FontImageDuration instead of letting battery meter just keep showing. 
#1 is a shift of an old problem where the effect image is stuck.
It used to be blst randomly sticking, and needing to do another effect like clash to make it stop.
Today after changes , it's reliably endlockup that's sticking.

profezzorn, 2021-11-10:
It's confusing to talk about different new problems in an old issue.
I'm closing this one, feel free to open a new one.

closed 2021-11-10 by profezzorn