profezzorn/ProffieOS #211: Feature Request: File-based blade styles

itsjohannawren, 2021-06-23:
I can hear the complaints about memory use and errors already, so hear me out.

Things that would need to be created:

* **A blade style profile file format** - Maybe just use the format from the [blade style editor](https://fredrik.hubbe.net/lightsaber/style_editor.html)
* **A blade style profile file name** - `blade.txt`?

Things that would need to be added:

* **A parser for the blade style profile file format** - This would be the hardest part because your not just taking a value from a line, you'd be creating a data structure that mirrors the structure in a file. It would have to know what's valid where in addition to knowing how to parse all the different styles and keep track of where it is in the structure.
* **A built-in `error` style to indicate a failure to parse a blade style profile file** - Dirt-simple `Blinking<Red,Black,500,500>`

Things that would need to be modified:

* **Style Search** - Add an attempt to load a file-based blade style profile before searching for a built-in style while loading a font. It should gracefully fall through to looking for a built-in style if no blade style profile file is found in the font directory. If it does find one but there's an error in it, the built-in `error` style should be returned.
* **Font Change** - A loaded file-based blade style profile should be unloaded (freed) when a font change is initiated.

Most of this is within my ability, but the parser is most definitely not. I'm honestly not sure where to even begin, especially since I write C, not C++ ;-) Definitely willing to learn, but I'd still need pointers on where to start.

itsjohannawren, 2021-12-08:
I was thinking about this more, @profezzorn. Is something like [Linux's execute in-place](https://www.electronicdesign.com/technologies/embedded-revolution/article/21805857/xip-with-linux-a-new-spin-on-embedded-architecture) available? IIRC, it effectively `mmap()`'s the sections of the binary to let them execute without actually being loaded into RAM. Granted, this functionality is really limited to flash memory with the right filesystem (not FAT), but it could be an option.

What I need to do is really dive into the code to figure this out.

profezzorn, 2021-12-08:
Short answer is no.
First of all, there is no mmu, so there is no mmap, or anything like it.
Second there is no real OS, so anything like this that we want to add, we have to write it ourselves.
It's not impossible to load code from SD and run it, but will have to be loaded into RAM, and it's not an easy thing to write.

itsjohannawren, 2021-12-09:
So much for that shortcut ;-)