profezzorn/ProffieOS #252: Font directory not found

jrice1701, 2021-09-26:
So i upgraded the sd card, copied and pasted the files and did a config in Arduino. Now in the serial monitor all the bank fies are there, but 0 nect to them. No sounds found. Whaat can i do?

profezzorn, 2021-09-26:
Seeing zeroes next to directories is normal when using the "dir" command.
Try the "pwd" command to see what font directory it's actually using. Does it match the names on the SD card?
Also try the "effects" command to see what sound sound files it found.

jrice1701, 2021-09-26:
I tried that but everything is corrupted. This is my 4th purchased board in
4 months. 600.00 just in stupid cores. Now it won't even accept the
standard 5 font proffie config. Web usb does nothing as well. I'm at a loss.

jrice1701, 2021-09-26:
I've lost all my data from the retailer. Now they want me to send it in and
I'll have to buy another one because I added a 28mm speaker and voided
their warranty. This is ridiculous. The blade is 3/4 of the length. The
power button only turns it on and I've lost all of my sa22c gesture files.
I'm lost. I keep reconfiguring to the standard one because that's all I
have.

profezzorn, 2021-09-26:
None of the information you've given me is helpful in actually figuring out what is going on though.
I need to know what you've done, what your config file looks like and the actual replies that you get from arduino or in the serial monitor. Saying "everything is corrupted" is just not helpful.

For webusb to work you need to select an option that includes webusb in arduino->tools->usb type before uploading.

closed 2021-10-06 by jrice1701

profezzorn, 2021-10-06:
This means that your config file is too large: C:\Users\Aarons\AppData\Local\Temp\arduino_build_51775/ProffieOS.ino.elf section .text' will not fit in region FLASH'

More information here: https://github.com/profezzorn/ProffieOS/wiki/Saving-Memory