profezzorn/ProffieOS #314: Power Off Blades for EFFECT_BATTERY_LEVEL and others

fdarosa2663, 2021-10-31:
Need to ensure blade powers off for layers using EFFECT_BATTERY_LEVEL, EFFECT_BOOT, EFFECT_NEWFONT, EFFECT_PREON, EFFECT_POSTOFF, EFFECT_LOW_BATTERY

profezzorn, 2021-10-31:
these three are handled:
EFFECT_BOOT,
EFFECT_PREON,
EFFECT_NEWFONT
here:
https://github.com/profezzorn/ProffieOS/blob/ce5c3e5c4dc9ec86accca46908f75eaeb5639c11/styles/transition_effect.h#L32

that leaves:
EFFECT_BATTERY_LEVEL
EFFECT_POSTOFF
EFFECT_LOW_BATTERY

EFFECT_POSTOFF might require a little special working since it CAN occur after retraction, but only once, so after it is done, it can't occur again and we can return ZERO_UNTIL_IGNITION.

The other two I don't when they can occur.
If they can occur after retraction, then we should probably make sure that they power the blade on, then we can handle them the same way as EFFECT_NEWFONT.

profezzorn, 2021-11-08:
I've made it so that all new effects turn the blade on.
This might not be the right thing to do for some effects, so we'll need to monitor and see if causes problems, but for now, this should be fixed.

closed 2021-11-18 by profezzorn