profezzorn/ProffieOS #328: OLED. No message shown if idle.bmp

NoSloppy, 2021-11-10:
If there's no font.bmp, the "message" from the preset does not show, it goes straight to idle.bmp.

NoSloppy, 2021-11-24:
I think I fixed this PR # 350

NoSloppy, 2021-11-25:
somewhat...

profezzorn, 2021-12-28:
I think this should be fixed now (https://github.com/profezzorn/ProffieOS/commit/b43fe0f2aa55dab9930191ae10cb299c05987dd5)
please try it and let me know if it works well for you.

NoSloppy, 2021-12-28:
Unfortunately, it seems no.
If there's no font.bmp, the "message" from the preset does not show, it goes straight to idle.bmp. (still true)

Also, new behavior if USB is connected. 
First, although "USB" is displayed, idle is allowed to play (although not displayed).
Serial Monitor shows "Playing idle.bmp" over and over again while it loops (unseen) and Mass Storage doesn't get a chance until switching to a preset whose font has no idle image.
Also, while USB still connected, (even if SD is mounted then unmounted from PC) once idle starts playing (masked by "USB") , next/prev preset does not show font.bmp. Just a blip of the idle image, then back to "USB".
However, if coming from a preset that has NO idle.bmp, OR if the current font.bmp playing is interrupted with next/prev preset before idle starts, the next/previous preset displays font.bmp fine.
So hard to put in words. 

Basically once we're in idle, next/prev skips font image.

If you want, I can make a video for clarity.

profezzorn, 2021-12-28:
The "Playing idle.bmp" turned out to be pretty much harmless. (Although quite annoying.)
Basically, it wouldn't actually play it, it just said it did.
However, I figured out the reason for the spamming and fixed it, so I think now the message will show up properly as well.

NoSloppy, 2021-12-28:
cool, things are better.
First, no crashing when SD card is busy is working great. Repeated attempts to next/previous/on/off all simply reply with Font Directory Not found, serial messages show NOT FOUND ....looking good.

Also, when USB not connected, all the things (font and idle images, messages) all show correctly.

However, the USB connected part is still a bit quirky.
Message still does not show if next/prev preset is done once idle starts playing. If cycling quickly through presets, switching before idle kicks in, everything displays properly.
If the current preset has no idle, next/prev preset will show whatever's next fine (message or font.bmp)

On boot, Mass Storage works and SD mounts to PC, regardless if preset has idle.bmp or not.
"USB" shows on screen if there's an idle image, and battery meter shows if no idle image. Great.
However, once USB is disconnected, reconnecting is not triggering an SD mount to PC.
It remains available to the Proffieboard (sounds play), and after switching to another preset, Serial Monitor is showing repeated
```
11:44:08.897 -> I2C init..
11:44:10.903 -> I2C sleeping..
11:44:13.899 -> I2C init..
11:44:15.921 -> I2C sleeping..
11:44:18.886 -> I2C init..
11:44:20.889 -> I2C sleeping..
11:44:23.894 -> I2C init..
11:44:25.917 -> I2C sleeping..
```

profezzorn, 2021-12-29:
Ok, third (?) time's a charm?
The repeated I2C messages are normal, and a consequence of using I2C intermittently.

NoSloppy, 2021-12-29:
Yes! All images and messages are working properly, nice work! (of course)

Now the only thing left is this USB reconnecting not remounting the SD, which was your #1 criteria at the beginning of all this idle business.
Do you want a dedicated issue generated for this "last hurdle" ?

profezzorn, 2021-12-29:
Hopefully it's just a matter of releasing the SD card faster.
I can have a go at that tomorrow.

NoSloppy, 2021-12-29:
I dropped this all the way down to 100 and no change.
[https://github.com/profezzorn/ProffieOS/blob/ea8b93d1d1298804f92fc8d6d8e81d9668392c0e/common/sd_card.h#L26](url)

profezzorn, 2021-12-29:
Hmm, I don't think that's fast enough.
However, I do wonder, because there has to be a way for something like a card reader to signal that media has been inserted, so there really should be a way for this to work, although it might require larger changes....

NoSloppy, 2021-12-30:
oh hey I made it 10 and it worked!

profezzorn, 2021-12-30:
Now about now?

NoSloppy, 2021-12-30:
> Now about now?

oops. Now it's mounting the SD to PC every preset switch :)

That's a lot of little edits added. Was there a reason it couldn't just be the shorter delay such as I tried above?

profezzorn, 2021-12-30:
I don't see mounting the SD card too often as a problem.

NoSloppy, 2021-12-30:
> I don't see mounting the SD card too often as a problem.
Oh, that one I have to beg to differ. That would be terribly annoying if it constantly remounted while switching presets with USB in. 
However, onto my follow up post I was typing while you answered there nd I was testing.

Stop the presses. Weird. It _was_ SD mounting with every next preset and wait to see PLI or idle.
Then I cycled quickly through a few presets, then it stopped. I thought maybe a MacOS fluke.
I rebooted the computer and and now the SD is not re-mounting even on reconnect USB.
I tried:
 - disconnect and reconnect USB while on a preset with PLI
 - disconnect and reconnect USB while on a preset with idle.bmp
 - disconnect, switch presets, then reconnect USB
 No luck. SD not re-mounting at all now.
 
 Sorry, I'm trying to streamline and be complete in testing, but now seems like back to square one.

profezzorn, 2021-12-30:
> Oh, that one I have to beg to differ.

Then re-flash without mass storage.

> It was SD mounting 

So what changed? Without knowing that I really can't do anything about it.
Maybe the SD card got corrupted somehow?

profezzorn, 2021-12-30:
Actually, re-mounting isn't possible.
It seems that you actually have to reboot the board for that.
(Although I should probably fix it so that disconnecting USB is enough.)
https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/ac659033eadd50cfe001ba1590a1362b2d87bb76/system/STM32L4xx/Include/dosfs_device.h#L68

Maybe when it was re-mounting for you, the board actually crashed?

NoSloppy, 2021-12-30:
Well that's certainly showing that anything I have ever had "working" were flukes indeed. 
No, the board showed no signs of crashing. In fact, when the SD was re-mounted, everything still showed normally like battery voltage every 20 secs, as well as `SD card is busy` messages and Font Directory Not found when doing next/prev preset etc... It simply behaved like the board had just booted.
Very weird.
Well, at least I can stop pulling out what hair I have left now.
What can I do now? Or moreso, what do you think should be a next step if any?

The goal was to have Mass storage work when USB connected.
I can confirm that works BOTH:
when the board is booted while disconnected -> then plugged in, 
and also if the board is connected ->then booted.
It's just that it's a single occurance of SD mounting until next boot? Which is....fine?
Does that satisfy the initial criteria?

profezzorn, 2021-12-31:
I think I'm going to call this good enough and close it for now I think.

closed 2021-12-31 by profezzorn