profezzorn/ProffieOS #383: TrRandom possibly bugged

NoSloppy, 2022-01-10:
Got a Squee of Death every time this this layer is used..
The board does not reset on power cycle. Still squee when power applied.
Needs manual bootloader->upload to come back to life.
I ruled out SD card to some extent, same results with different fresh cards, 
however the board does say font directory not found/SD card not found when the SD is not installed.
I tried different fonts as well and any one does it, which led me to blade style.

The best I can narrow this down to is having a TrRandom in a TrasitionLoop.
```
    AlphaL<
        PulsingL<
            TransitionLoop<
                AlphaL<ColorSequence<3500,Cyan,Black,Rgb<25,10,255>,Black>,Int<8000>>,
                TrRandom<
                    TrConcat<TrBoing<500,2>,Black,TrInstant>,
                    TrJoinR<
                        TrJoin<TrWipeInX<Int<300>>,TrWaveX<White,Int<400>,Int<600>,Int<300>,Int<32768>>>,
                        TrJoin<TrWipeX<  Int<300>>,TrWaveX<White,Int<400>,Int<600>,Int<300>,Int<0>>>>>>,
        Int<1500>>,
    Int<16000>>,
```

Having a different transition there, everything works fine:
```
          AlphaL<
              PulsingL<
                  TransitionLoop<
                      AlphaL<ColorSequence<3500,Cyan,Black,Rgb<25,10,255>,Black>,Int<8000>>,
                      TrConcat<
                          TrBoing<500,2>,Black,
                          TrJoinR<TrWipeInSparkTipX<White,Int<300>>,
                          TrWipeSparkTipX<White,Int<300>>>>>,
              Int<1500>>,
          Int<10000>>,
```

profezzorn, 2022-01-11:
Looks like TrRandom triggers it, but the bug is actually in TransitionLoop.
I think it should be fixed now.

NoSloppy, 2022-01-11:
Works now. great!

profezzorn, 2022-01-11:
Guess we'll have a 6.5 soon. :)

closed 2022-01-11 by profezzorn