profezzorn/ProffieOS #391: pov_tools working on themselves

NoSloppy, 2022-01-17:
On both Windows with WSL and MacOS, the files:
pgmtorle.cc
pnmquantizedtorle.cc
pnmtorle.cc
and pnmwindshieldwiper.cc
are having hex versions of themselves created in the directory alongside themselves.
Additionally, on MacOS, due to Developer Tools, the -b debugging flag for g++ is creating ELF files with DWARF information in the form of the files*.dSYM packages.

profezzorn, 2022-01-17:
If those files are needed for debugging, I would rather have them than not.
Are they a problem?
(PS: I assume you mean the -g option.)

NoSloppy, 2022-01-17:
I wouldn't say a problem. Just an unexpected clutter when there's now three copies of each file. Does not really matter. I just thought I would report it as unexpected. Really the hex files more so than the debugging files.

NoSloppy, 2022-01-26:
Closing. It's been determined the tools are uncompiled initially, and the generated hex versions are the compiled programs themselves. Removing them will only mean needing to recompile and slow things down.

closed 2022-01-26 by NoSloppy