profezzorn/ProffieOS #449: Feature Request: SubBlade with manual defined pixel inclusion

Afrobaron, 2022-09-12:
Currently 
- SubBlade allows define of a sequential set of pixels in a specified amount.  
- SubBladeWithStride allows skipping a set number of pixels to allow mathematic allocations of non-sequential pixels.

This feature would allow a manual pixel definition that cannot be defined with a stride.  {1,6,7,12,13,18,19,24,25} {3,4,9,10,15,16,21,22}

Reasoning for feature need:
Pixels when grouped closely on a single PCB in a matrix are easier to zigzag the data line.  The attached picture shows a pixel matrix and shows what a stride by 3 looks like vs. what this feature request would allow to be created.

[image]

profezzorn, 2022-09-12:
This is an interesting special case where every other "stride" is backwards, which would be fairly easy to implement I think. Is there an actual PCB that uses this?

Afrobaron, 2022-09-12:
There is a PCB available today.  It is meant as a pixel replacement to a PLI/Bar Graph.  SubBlade combined with SubBladeReverse can get 14 individual rows of three pixels for a great bar graph effect, but I am looking to go the opposite way.

[ShadowFoil's PCB](https://shadowfoil.com/products/hangman-pcb)

Afrobaron, 2022-09-12:
Thinking about your stride with reverse feature, it may not work for the red colored blade.  

Red blade - Stride one (forward) would start on 3, then stride, pickup the pixel , then reverse stride.  This would mean pixel 4 would not get picked up. 

You are the wizard on this part but I would figure picking up two pixels in a row after a stride or reverse stride would help.  However, for the first blade picking up two pixels at the start on the very first pixel would not be correct either.  I think a manual defined array of pixel numbers might be a way to go.

SubBladeDefined({array_of_int}, {array size}, blade_definition)
SubBladeDefined({1,6,7,12,13,18,19,24,25,30,31,36,37,42}, 14, WS281XBladePtr<42, bladePin, Color8::GRB, PowerPINS<bladePowerPin2)> >())

profezzorn, 2022-09-12:
Something like SubBladeZZ (ZZ standing for ZigZag) can work, but it would have to be defined differently than SubBladeWithStride, something like:

SubBladeZZ(start_of_stride, number_of_strides, length_of_stride, pixel_within_stride, BLADE);

So for blue it would be:
SubBladeZZ(0, 14, 3, 0, WS281XBladePtr<42, bladePin, Color8::GRB, PowerPINS<bladePowerPin2)> >());

green:
SubBladeZZ(0, 14, 3, 1, NULL);

red:
SubBladeZZ(0, 14, 3, 2, NULL);

If done correctly, it would also allow for negative strides, which would cover the "reverse" part.

Afrobaron, 2022-09-12:
If I get your example correctly, _length_of_stide_ = 3 would only put us in the same position we are in currently.  
I'm not sure I follow the use of the _pixel_within_stride_ purpose/function.

In reflection this does make the end user have to do a fair amount of head math and paper mapping that could be avoided if the input was an array of int.

profezzorn, 2022-09-12:
pixel_within_stride is just which pixel in the first row you want to use.
Seems pretty simple to me.

Afrobaron, 2022-09-12:
If that is the case your example:
> SubBladeZZ(0, 14, 3, 0, WS281XBla.....

1. Will divide the overall sequence of 42 pixels into 3 pixel sections (14 sections total).
 - Each section will be as examples {1,2,3}, {6,5,4}, {7,8,9}, etc. 
2. Then it will add to the blade the first pixel from each section - 1,6,7,....

Am I following correctly?  If yes, then this is simple enough for end users to understand and use.

profezzorn, 2022-09-12:
Yes, that is exactly how I was thinking it would work.

Afrobaron, 2022-09-12:
Possible OS7 feature or a late 6.x ?

Afrobaron, 2022-11-13:
@profezzorn did this request make it into OS7 alpha stage?  I would like to test it out and since I have the hardware on hand should be fairly easy to try.

profezzorn, 2022-11-20:
SubBladeZZ is now checked in.

Afrobaron, 2022-11-21:
Took me some fiddling around on the config.  Your final implementation and the one above are slightly different. Above you used number of strides, the final implementation uses total LED count.  A quick scan of the code let me find it.

Here is a demo video - [YouTube Video Demo](https://youtu.be/zXoHwEVY0rE)
first blade (green) is a standard single strip blade.  The three other blades (red, white, blue) all used SubBladeZZ

`{ 0,
	WS281XBladePtr<10, bladePin, Color8::GRB, PowerPINS<bladePowerPin3> >(),
	SubBladeZZ(0, 41, 3, 0, WS281XBladePtr<42, blade2Pin, Color8::GRB, PowerPINS<bladePowerPin4>>()),
	SubBladeZZ(0, 41, 3, 1, NULL),
	SubBladeZZ(0, 41, 3, 2, NULL),
CONFIGARRAY(presets) },`

closed 2022-12-29 by profezzorn