profezzorn/ProffieOS #474: ssd1306 new weird behavior; matching files

NoSloppy, 2022-11-01:
The current 6.7 version tested and compared to the master before any PR adds, and then including pending PR edits. Same results through (So it's as of 6.7)
 
If image effects have digits like blst01.bmp, they’re unexpectedly playing 1:1 with the numerical wav counterpart, similar to a paired effect like preon->out, or matching a COLOR if using ColorSelect<WavNum <>.

If the quantity of numbered image files > quantity of wavs, the un-matched image never shows.
However,
If the quantity of wavs exceeds the quantity of numbered image files, it causes OLED temporary freeze with either 
‘read image fail’ repeating,
or 
`Unknown image format. a=-1 b=-1 pos=0
read image fail` repeating.

Finally, if there’s only a single unnumbered image file, such as blst.bmp, things work fine regardless of quantity of wavs.

But, a mix of one unnumbered and numbered files (blst, blst02, blst03) The same freeze/read fail occurs.

I extensively tested all scenarios including subdirectories or not, single frame or looped bmp files, etc… same results.

profezzorn, 2022-11-01:
This is because of a change I made DoEffect<> to let DoEffect select which file to use.
I'll see what I can do about it...

profezzorn, 2022-11-01:
Hopefully this should be fixed now. Don't have time to test it properly right now.

NoSloppy, 2022-11-01:
I see. Will test, thanks.

NoSloppy, 2022-11-01:
Looking good, all working. These results are what you expected?
Effect is set to paired using ProffieOS.IMG.EFFECTNAME.paired=1. 
If quantity of images=wavs, all files are paired 1:1.
If quantity of images<wavs,  higher numbered wavs all use highest numbered image.
If quantity of images>wavs, unmatched   images get ignored.

Not set to paired, images are randomly chosen, regardless of quantities of sounds/images.

profezzorn, 2022-11-17:
excellent

closed 2022-11-17 by profezzorn