profezzorn/ProffieOS #475: OLED imaged unknown image format, but working

NoSloppy, 2022-11-01:
Looping stacked frame BMP diplays fine, no error.
Single frame BMP displays fine, but `Unknown image format. a=-1 b=-1`
non-looped PBM also works but `Unknown image format. a=-1 b=-1`
Seems they're just not read right and not matching a case in the TAG2 switch?
Looking at the PBM file, it does have a "P4" magic number.

For non-looped options, only PBM is working by using Imagemagick `convert` 
Just doing ‘cat *bmp > boot.bmp’ makes file, but doesn’t animate on OLED. Shows first frame only for duration.
Windows ‘copy /b *bmp boot.bmp’ =  Same results.

It would be great if a 3rd party command line image tool wasn't required to get these non-looping to work.
It's a bunch of hoops for the average user to get it installed with NetPBM.

profezzorn, 2022-11-17:
Could you attach the files that have the problem(s)?

NoSloppy, 2022-11-17:
> Single frame BMP displays fine, but 'Unknown image format. a=-1 b=-1'

https://www.dropbox.com/s/x073dbds3fyeg3q/out.bmp?dl=1

>Non-looped PBM also works but 'Unknown image format. a=-1 b=-1'
Seems they're just not read right and not matching a case in the TAG2 switch?
Looking at the PBM file, it does have a 'P4' magic number, yet the error gets reported.

https://www.dropbox.com/s/o92wfchcc3o2qhy/preon.pbm?dl=1

> For non-looped options...........(attempting BMP. Again, PBM works fine on screen, although reports Unknown format))
> Just doing `cat *bmp > boot.bmp` makes a file, but it doesn’t animate on OLED. Shows first frame only for duration.

Actually, today it's not showing anything, and also has `Unknown image format. a=-1 b=-1 pos=62130`
https://www.dropbox.com/s/24egu08nqlu6i3c/clsh.bmp?dl=1

profezzorn, 2022-11-17:
I'm guessing that the "unknown image format" things just happen when it reads past the end of the file.
Might not be worth fixing, but I will take a look.

profezzorn, 2023-03-09:
This is still an issue, right?
(If so, I should take a look at it before 7.x comes out of beta.)

NoSloppy, 2023-03-10:
It is. It does it, and to my best recollection, I couldn't narrow it down to the file format,  or processing steps, or any traceable attribute.
I'm up for trying any tests to help.

NoSloppy, 2026-10-24:
Closing. PR #900 addresses it.

closed 2026-10-24 by NoSloppy