profezzorn/ProffieOS #532: Save mem by omitting printouts

NoSloppy, 2023-02-19:
Scour the OS for omittable messages that can be turned on/off via define for debugging, and check the memory savings.

bmartinson, 2023-07-09:
@NoSloppy I'm currently going through this process for the files that I'm working with for my saber build. In my iteration, I am piggybacking off of the `DEBUG` define for this. If I make a PR to help support this effort, do you like that approach or do you think we should use a different compiler directive define to disable stdout usage seperate from `DEBUG`?

NoSloppy, 2023-07-09:
I'm not sure. I think a test is in order first to just see what the total used memory is. I have a feeling it's not going to be significant enough to bother with?
My gut reaction is to just blanket them under DISABLE_DIAGNOSTIC_COMMANDS maybe? 
But of course some discernment is required. Like, Error messages should probably remain.

profezzorn, 2023-07-09:
I really want to make a VLOG(N) macro for this.
The idea is that you just write something like:

  VLOG(5) << "Something happened.\n";

Then we have a define like VLOG_LEVEL, and all log lines with higher numbers just gets compiled out.
Ideally I also want some way to set vlog levels on a per file basis, so that maybe you can do something like:

#define VLOG_WS2811_H 100

and then all vlog levels in the ws2811.h file gets enabled.

bmartinson, 2023-07-09:
Sounds reasonable to me. While I haven't scoured the entire repository, I looked at a handful of files to remove direct STDOUT statements or general conditional blocks or loops that were solely for the purpose of writing to standard output. I've done so on the latest stable 7.8 release (so not latest in master).

I've been able to shave off 4272 bytes doing so which has been enough to meet the goal of getting one final blade style flashed onto my board. I'll share the details of which files I touched and for how much savings to serve as a spike into the potential outcome of working on this:

| File | Savings (bytes)     |
|--------------|-----------|
| ProffiOS.ino | 488 |
| blades/ws2811_blade.h | 224 |
| props/prop_base.h | 1768 |
| props/saber_BC_buttons.h | 752 |
| sound/effect.h | 744 |
| sound/hybrid_font.h | 232 |
| sound/sound.h | 64 |

NoSloppy, 2023-07-09:
> I've been able to shave off 4272 bytes doing so

Wow, surprising.

profezzorn, 2023-07-10:
Ok, so I've added a PVLOG() function which should make it easy to have different levels of logging.
You can find it common/stdout.h, and there are examples of how to use it in props/prop_base.h

Please note that PVLOG() is not suitable for all printouts.
Essential prinouts, or respones to serial monitor commands should not use PVLOG()
PRs that convert existing log statements to use PVLOG are welcome, but please keep the PRs relatively small, I don't want one mega-PR which converts thousands of logging statements in one go.

profezzorn, 2023-12-08:
I'm closing this issue, PVLOG is the solution, even if far from all printouts have been converted to PVLOG yet.

closed 2023-12-08 by profezzorn