profezzorn/ProffieOS #557: i swear i followed guides. pc wont recognize proffieboard

mitchell999901, 2023-05-17:
i got my first proffie prebuilt so idk what version its running. its a 2.2 thats all i know
i followed 5 guides all say the same thing up to this point. i have arduino i added the url to preferences changed board to proffie v2 tried os v2.2 and 3.6 tools board in arduino is 2.2 usb type is serial mass storage. one guide mentioned the port type but my arduino doesnt give me that option and says no proffie connected in the bottom right  
i opened arduino from proffie os v6.9 and it shows a whole bunch of code.
go to zadig and even with pressing the boot and resets it doesnt show up as stm32 bootloader all i have is serial under usb serial bus devices. 

i tried http://fredrik.hubbe.net/lightsaber/webusb.html#troubleshooting the usb interface but it says no web user interface found
tried 2 seperate micro usbs on 3 different usb ports  
on windows 11 and my devices shows proffieboard  and i can eject it but nothing else recognizes it or the sd card plugged in. my sd card is under the board and looks like it has no clearance to remove it. and im not sure what is done on sd card vs board
idk if i need to do all the steps since it already has some version of proffie os but i cannot access any files

profezzorn, 2023-05-17:
Most of the time, this turns out to be bad or charge-only cables.
I think the record so far is 5 bad cables, and the 6th one worked.
Try go to amazon and look for an USB-IF certified micro-usb cable, that usually helps.

https://pod.hubbe.net/troubleshooting/usb-connection-issues.html

closed 2023-05-17 by profezzorn

mitchell999901, 2023-05-20:
i got https://www.amazon.com/gp/product/B00UUBRX0Y/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1 this but no change. in arduino it says no ports detected. zadig still just says serial. its definitely the proffie usb as it only shows up when plugged in .  
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:53:2: error: #error Please set CONFIG_FILE as shown above.
   53 | #error Please set CONFIG_FILE as shown above.
      |  ^~~~~
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:57:10: error: #include expects "FILENAME" or <FILENAME>
   57 | #include CONFIG_FILE
      |          ^~~~~~~~~~~
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:610:10: error: #include expects "FILENAME" or <FILENAME>
  610 | #include CONFIG_FILE
      |          ^~~~~~~~~~~
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:614:10: error: #include expects "FILENAME" or <FILENAME>
  614 | #include CONFIG_FILE
      |          ^~~~~~~~~~~
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:618:10: error: #include expects "FILENAME" or <FILENAME>
  618 | #include CONFIG_FILE
      |          ^~~~~~~~~~~
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:676:10: error: #include expects "FILENAME" or <FILENAME>
  676 | #include CONFIG_FILE
      |          ^~~~~~~~~~~
C:\Users\mitch\AppData\Local\Programs\Arduino IDE\ProffieOS-v7.7\ProffieOS\ProffieOS.ino:1554:10: error: #include expects "FILENAME" or <FILENAME>
 1554 | #include CONFIG_FILE
      |          ^~~~~~~~~~~

exit status 1

Compilation error: #error Please set CONFIG_FILE as shown above. 
 i try and set my config like how i saw in one tutorial but it doesnt work

profezzorn, 2023-05-20:
This particular error just means that no CONFIG_FILE define was set in ProffieOS.ino
Once you've added/edited your CONFIG_FILE define, make sure it's not commented out.
(If it has // at the beginning of the line, you need to remove that.)

profezzorn, 2023-05-20:
The cable you have *should* work, but micro-usb cables are notorious for not working well, so it is a good idea to try a different cable.