profezzorn/ProffieOS #558: Unknown Error Code

Tipicole, 2023-05-17:
I was updating a blade style for my Rey font and pasted the new style code in (from OS6 Library) but got this error and don't know what it means

Arduino: 1.8.19 (Windows 10), Board: "Proffieboard V2, Serial + Mass Storage + WebUSB, SDCARD (SPI), 80 MHz, Smallest Code"

In file included from E:\The Files\Lightsaber\ProffieOS\ProffieOS.ino:563:

E:\The Files\Lightsaber\ProffieOS\config\maincon2.h:205:3764: error: wrong number of template arguments (2, should be 3)

  205 | StylePtr<Layers<Mix<HoldPeakF<SwingSpeed<250>,Scale<SwingAcceleration<100>,Int<50>,Int<500>>,Scale<SwingAcceleration<>,Int<20000>,Int<10000>>>,RandomFlicker<StripesX<Int<14000>,Scale<HoldPeakF<SwingSpeed<200>,Scale<SwingAcceleration<100>,Int<50>,Int<300>>,Scale<SwingAcceleration<100>,Int<24000>,Int<16000>>>,Int<-4000>,Int<-400>>,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>,Mix<Int<7710>,Black,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>>,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>,Mix<Int<19276>,Black,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>>>,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>>,RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>>,TransitionEffectL<TrConcat<TrJoin<TrDelayX<WavLen<>>,TrWipeIn<200>>,AlphaL<RgbArg<STAB_COLOR_ARG,Rgb<255,255,255>>,SmoothStep<IntArg<MELT_SIZE_ARG,28000>,Int<2000>>>,TrWipe<200>>,EFFECT_STAB>,TransitionEffectL<TrWaveX<RgbArg<BLAST_COLOR_ARG,Rgb<255,255,255>>,Scale<EffectRandomF<EFFECT_BLAST>,Int<100>,Int<400>>,Int<100>,Scale<EffectPosition<EFFECT_BLAST>,Int<100>,Int<400>>,Scale<EffectPosition<EFFECT_BLAST>,Int<28000>,Int<8000>>>,EFFECT_BLAST>,TransitionEffectL<TrConcat<TrInstant,AlphaL<RgbArg<CLASH_COLOR_ARG,Rgb<255,255,255>>,Bump<Scale<BladeAngle<>,Scale<BladeAngle<0,16000>,Sum<IntArg<LOCKUP_POSITION_ARG,16000>,Int<-12000>>,Sum<IntArg<LOCKUP_POSITION_ARG,16000>,Int<10000>>>,Sum<IntArg<LOCKUP_POSITION_ARG,16000>,Int<-10000>>>,Scale<ClashImpactF<>,Int<12000>,Int<60000>>>>,TrFadeX<Scale<ClashImpactF<>,Int<200>,Int<400>>>>,EFFECT_CLASH>,LockupTrL<AlphaMixL<Bump<Sin<Int<12>,Sum<IntArg<LOCKUP_POSITION_ARG,16000>,Int<-8000>>,Sum<IntArg<LOCKUP_POSITION_ARG,16000>,Int<8000>>>,Scale<SwingSpeed<100>,Int<14000>,Int<18000>>>,AudioFlickerL<RgbArg<LOCKUP_COLOR_ARG,Rgb<255,255,255>>>,BrownNoiseFlickerL<RgbArg<LOCKUP_COLOR_ARG,Rgb<255,255,255>>,Int<200>>>,TrConcat<TrInstant,RgbArg<LOCKUP_COLOR_ARG,Rgb<255,255,255>>,TrFade<400>>,TrConcat<TrInstant,RgbArg<LOCKUP_COLOR_ARG,Rgb<255,255,255>>,TrFade<400>>,SaberBase::LOCKUP_NORMAL>,LockupTrL<AlphaL<RgbArg<LB_COLOR_ARG,Rgb<255,255,255>>,LayerFunctions<Bump<Scale<SlowNoise<Int<2000>>,Int<3000>,Int<16000>>,Scale<BrownNoiseF<Int<10>>,Int<14000>,Int<8000>>>,Bump<Scale<SlowNoise<Int<2300>>,Int<26000>,Int<8000>>,Scale<NoisySoundLevel,Int<5000>,Int<10000>>>,Bump<Scale<SlowNoise<Int<2300>>,Int<20000>,Int<30000>>,Scale<IsLessThan<SlowNoise<Int<1500>>,Int<8000>>,Scale<NoisySoundLevel,Int<5000>,Int<0>>,Int<0>>>>>,TrConcat<TrInstant,AlphaL<RgbArg<LB_COLOR_ARG,Rgb<255,255,255>>,Bump<Int<12000>,Int<18000>>>,TrFade<200>>,TrConcat<TrInstant,HumpFlickerL<AlphaL<RgbArg<LB_COLOR_ARG,Rgb<255,255,255>>,Int<16000>>,30>,TrSmoothFade<600>>,SaberBase::LOCKUP_LIGHTNING_BLOCK>,LockupTrL<AlphaL<RandomPerLEDFlickerL<RgbArg<DRAG_COLOR_ARG,Rgb<255,255,255>>>,SmoothStep<IntArg<DRAG_SIZE_ARG,28000>,Int<3000>>>,TrConcat<TrJoin<TrDelay<4000>,TrWipeIn<200>>,AlphaL<BrownNoiseFlickerL<RgbArg<DRAG_COLOR_ARG,Rgb<255,255,255>>,Int<300>>,SmoothStep<IntArg<DRAG_SIZE_ARG,28000>,Int<3000>>>,TrFade<4000>>,TrFade<300>,SaberBase::LOCKUP_DRAG>,LockupTrL<AlphaL<Mix<TwistAngle<>,RgbArg<STAB_COLOR_ARG,Rgb<255,24,0>>,RotateColorsX<Int<3000>,RgbArg<STAB_COLOR_ARG,Rgb<255,24,0>>>>,SmoothStep<IntArg<MELT_SIZE_ARG,28000>,Int<4000>>>,TrConcat<TrWipeIn<100>,AlphaL<RgbArg<STAB_COLOR_ARG,Rgb<255,24,0>>,SmoothStep<IntArg<MELT_SIZE_ARG,28000>,Int<4000>>>,TrFade<300>>,TrWipe<200>,SaberBase::LOCKUP_MELT>,InOutTrL<TrWipeSparkTipX<RgbArg<IGNITION_COLOR_ARG,Rgb<255,255,255>>,IgnitionTime<300>>,TrWipeInSparkTipX<RgbArg<RETRACTION_COLOR_ARG,Rgb<255,255,255>>,RetractionTime<0>>,ColorSelect<IncrementWithReset<EffectPulseF<EFFECT_IGNITION>,EffectPulseF<EFFECT_NEWFONT>,Int<1>>,TrInstant,Mix<SmoothStep<IntArg<EMITTER_SIZE_ARG,2000>,Int<-1000>>,Black,BrownNoiseFlicker<RgbArg<BASE_COLOR_ARG,Rgb<255,255,0>>,Black>,200>>,Black>>,TransitionEffectL<TrConcat<TrJoin<TrDelayX<Int<100>>,TrInstant>,AlphaL<Green,SmoothStep<IntArg<PREON_SIZE_ARG,2000>,Int<-2000>>>,TrWipe<100>,AlphaL<DodgerBlue,SmoothStep<IntArg<PREON_SIZE_ARG,2000>,Int<-2000>>>,TrDelayX<Sum<WavLen<EFFECT_PREON>,Int<200>>>>,EFFECT_PREON>>>(),"rey"},

      |                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    ^

In file included from E:\The Files\Lightsaber\ProffieOS\ProffieOS.ino:432:

[... cut ...]                                                                                                                                                                                                                                                                                                                           ^

exit status 1

Error compiling for board Proffieboard V2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

profezzorn, 2023-05-17:
I don't know what happened, but obviously this style doesn't work as written.
It seems like someone has mixed up BrownNoiseFlicker and BrownNoiseFlickerL, and there may be additional problems as well.

closed 2023-05-17 by profezzorn

Tipicole, 2023-05-17:
Very odd, i thought it was a coding mistake on my part. maybe its just a bug with the Dual Phase option in the editor?

profezzorn, 2023-05-17:
Unless you somehow changed it while cut-n-pasting, that would be my guess.