profezzorn/ProffieOS #649: sound library is saying "nine" instead of "zero"

NoSloppy, 2024-04-04:
I didn't revert to see when this started, and don't see anything glaring.
Sending `batt` in serial monitor speaks "three point nine two volts" and displays 3.92.
Triggering spoken battery level with prop (nothing changed lately in that feature),
```cpp
          sound_library_.SayNumber(battery_monitor.battery(), SAY_DECIMAL);
          sound_library_.SayVolts();
```
speaks "three point zero two volts" and displays 3.92.
I confirmed that mzero.wav and mnum9.wav are correct.

profezzorn, 2024-04-04:
I think it should be fixed now.

NoSloppy, 2024-04-04:
Not quite. 3.68 is "three point thirty six eight volts"

profezzorn, 2024-04-04:
How about now?

NoSloppy, 2024-04-04:
now it doesn't speak at all

NoSloppy, 2024-04-04:
now it works.

closed 2024-04-04 by NoSloppy