profezzorn/ProffieOS #651: ToggleColorChangeMode broken

NoSloppy, 2024-04-04:
I was testing changes to my prop and thought I was crazy.
Calling ToggleColorChangeMode() enters fine, but no controls intended to exit work.
Conversely, the oddest thing is that `EVENT: Power-SavedShortclick#1` plays ccend.wav and exits CC mode, but that is only set to do a blaster block in my code.

Even weirder, clicking AUX exited the mode:
```cpp
03:57:47.051 -> EVENT: Aux-SavedShortclick#1 ON millis=61363
03:57:47.051 -> unit = 1 vol = 0.50, Playing common/ccend.wav
```
yet `EVENTID(BUTTON_AUX, EVENT_FIRST_SAVED_CLICK_SHORT, MODE_ON)` does not even exist anywhere in the prop file at all.

I put a printout directly after the click POW event:
```cpp
      case EVENTID(BUTTON_POWER, EVENT_FIRST_SAVED_CLICK_SHORT, MODE_ON):
STDOUT << "*** Why is this exiting ColorChange mode???\n";
```
and it does not print if in CC mode. 
```cpp
04:11:52.830 -> EVENT: Power-SavedShortclick#1 ON millis=45179
04:11:52.830 -> unit = 1 vol = 0.50, Playing common/ccend.wav
```
it prints fine if NOT in CC mode.
```cpp
04:00:44.651 -> EVENT: Power-SavedShortclick#1 ON millis=238967
04:00:44.651 -> *** Why is this exiting ColorChange mode???
```
Lastly, I tried from Serial Monitor, and sending `ccmode` enters the mode, but I got a slew of Save file notices:
```cpp
04:11:01.056 -> Creating file 00_2ndSisterSave/presets.ini iteration = 249
04:11:01.241 -> Creating file 00_2ndSisterSave/presets.tmp iteration = 250
04:11:01.277 -> Creating file 00_2ndSisterSave/presets.ini iteration = 251
04:11:01.346 -> Creating file 00_2ndSisterSave/presets.tmp iteration = 252
04:11:01.417 -> Creating file 00_2ndSisterSave/presets.ini iteration = 253
04:11:01.489 -> Creating file 00_2ndSisterSave/presets.tmp iteration = 254
```
that just kept going.

profezzorn, 2024-04-04:
Some of this is definitely "working as intended".
When in a mode, all events are sent to the mode instead of the prop, and it's up to the mode to decide when to exit from the mode, so the prop doesn't have to worry about it.   This should also reduce the number of  "if (!menu_) return" in the prop code.

That said, all of this is new code, and there are probably a bunch of bugs, rough edges and things that simply could work better.....

NoSloppy, 2024-04-04:
So we're losing the ability to control exiting with a designated button action in the prop?
At minimum this breaks being able to revert and exit without saving changes .

profezzorn, 2024-04-05:
Hmm, this was the first mode I wrote, so I might have made it wrong.
The way it's supposed to work is that the one button exits and saves and another button exits and reverts.

It's still possible to change which button does what, but doing so is done differently than it was done in the past. If you look at ToggleColorChangeMode you'll see a reference to COLOR_CHANGE_MENU_SPEC_TEMPLATE, which is defaulted to ColorChangeOnlyMenuSpec a few lines above.  ColorChangeOnlyMenuSpec is defined at the end of modes/color_change_modes.h

The very first entry in ColorChangeOnlyMenuSpec is SelectCancelMode, which is the part of the code that defines which button does what.  The default SelectCancelMode is defined in modes/select_cancel_mode.h, however, you can absolutely replace the default SelectCancelMode with something else.

You could write a class in your prop file which inherits from SelectCancelMode, but overrides mode_Event2 to do whatever you want.  Then you make a SPEC which inherits from ColorChangeOnlyMenuSpec, but overrides SelectCancelMode to use your class. Finally, override ToggleColorChangeMode, or change the COLOR_CHANGE_MENU_SPEC_TEMPLATE define so that your spec will be used instead of the default one.

I guess there are a few more steps than there used to be, but in return, we get modularity, each mode is defined in it's own class, and we don't need to worry about it in the prop class. Anyways, hope all of this makes sense, it's not every day that you have to explain how to do something in a programming paradigm that didn't exist before.... (AFAIK)

NoSloppy, 2024-04-05:
> it's not every day that you have to explain how to do something

Well it's much appreciated and certainly explains the behavior. I guess it's time to dive into the modes folder, as I've been ignoring it until you were "done" and there was maybe a doc about how it all works.

profezzorn, 2024-04-05:
For now, you should probably ignore everything except the color change modes.
There is a lot of errors and problems in the rest of the code, and I'm working on fixing that.

NoSloppy, 2024-04-05:
Got it. Wouldn't this break backwards compatibility though as far as existing prop files that already handle saving or reverting color change for example?

NoSloppy, 2024-04-05:
Also, is ZOOMED going to be included in the modes? I had been considering adding that to my prop.
All that depends on how we're planning to handle button events though. If moving to a universalized system, I'll hold off to see what that becomes. Although I still wonder on compatibility with existing versions of the prop. 
Is requiring a prop version to match the OS version a thing?

profezzorn, 2024-04-06:
My current idea is to use 3 revolutions to get back to the original color, and skip having a "Zoomed" mode.
I'm not sure how well that will work though, we'll just have to try it and see.
Again, the whole point of this system is to let people customize, so if you want zooming, then zooming you can have. :)

And yes, props are not guaranteed to work with other version of the OS. They sometimes do, and even when they don't it's usually not particularly difficult to fix. Usually when I make incompatible changes in the OS, I update all the props to work with it right away. But that means that props that live outside of the ProffieOS code base still needs updating.

profezzorn, 2024-04-06:
All the code is checked in now. However, I haven't tested any of it, so I'm sure it's got more bugs than an anthill, and there is a good chance that some things will need to be changed/fixed/added, but it all compiles, and hopefully the general principles of how it works are good. I'm going to Oregon for a week, so I'm not sure if I'll be doing anything with this stuff until I get back.

fdarosa2663, 2024-04-30:
Haven't really been able to keep up with the changes, if these changes are impacting how ToggleColorChangeMode is working in my prop, how do we prevent or "go backwards" to how it was working previously?   It was reported that the buttons aren't working in current version but I haven't changed anything.

profezzorn, 2024-04-30:
If you really want it to work the way it did before, then you'll need to copy the old color change mode code into your prop, including ToggleColorChangeMode().

Basically, reverse this change:
https://github.com/profezzorn/ProffieOS/commit/cf85bd8a425af533d3628ebac0c00e142d5632fa
(but in your prop)

fdarosa2663, 2024-05-01:
Thanks, noted.  I'll have a go at it when I get a chance.

NoSloppy, 2024-06-19:
So what is the difference between select_cancel_mode and accept_cancel_mode, aside from the serial monitor commands option?

profezzorn, 2024-06-19:
A very good question. Looks like some accidental duplication.
accept_cancel_mode.h is not actually included anywhere though, I will delete it.

NoSloppy, 2024-06-19:
Based on you direction/explanation above, does this look right to customize the color change controls in my prop?
Seems to be working well.
```cpp
// Custom SelectCancelMode class
  class BCSelectCancelMode : public mode::SelectCancelMode {
  public:
    BCSelectCancelMode() {}

    bool mode_Event2(enum BUTTON button, EVENT event, uint32_t modifiers) override {
      switch (EVENTID(button, event, 0)) {
        case EVENTID(BUTTON_POWER, EVENT_FIRST_HELD_MEDIUM, 0):
          // Save and exit
          select();
          return true;

        case EVENTID(BUTTON_POWER, EVENT_SECOND_SAVED_CLICK_SHORT, 0):
          // Exit without saving
          exit();
          return true;

        case EVENTID(BUTTON_POWER, EVENT_THIRD_SAVED_CLICK_SHORT, 0):
          // Revert to original color (reset variation)
          SaberBase::SetVariation(0);
          PVLOG_NORMAL << "Revert Color Variation to uploaded config color. Variation = " << SaberBase::GetCurrentVariation() << "\n";
          SaberBase::SetColorChangeMode(SaberBase::COLOR_CHANGE_MODE_NONE);
          popMode();
          return true;
      }
      return false;
    }
  };

// Custom Color Change Menu Spec
template<class SPEC>
struct BCColorChangeOnlyMenuSpec : public ColorChangeOnlyMenuSpec<SPEC> {
  typedef BCSelectCancelMode SelectCancelMode;
};
```

profezzorn, 2024-06-19:
Most of this looks exactly right, except for one thing...
If it compiles and works, it's probably fine, but the revert-to-original button doesn't really belong in the SelectCancelMode.
In a larger menu system, almost all menus inherit from SelectCancelMode, so color-reverting doesn't really make sense there. Unfortunately SteppedVariantionMode and SmoothVariationMode don't share a common subclass, so if you wanted to do it the "right" way, you would need to do the revert thing twice.

Regarding (1) and (2): That is kind of the point... :)

NoSloppy, 2024-06-19:
Oh.. duh. I could just have a standalone function to revert, there's no need to do it while in or when exiting CC mode. 
Ok cool .
Fiddling with using the smooth volume right now .

closed 2024-07-03 by NoSloppy