profezzorn/ProffieOS #662: latest update bug: ServoSelector

NoSloppy, 2024-06-10:
```cpp
ProffieOS/blades/simple_blade.h:316:31: error: 'ServoSelector' does not name a type
  316 | template<int pin, class LED = ServoSelector>
      |                               ^~~~~~~~~~~~~

```

profezzorn, 2024-06-10:
Hmm, I guess I forgot to check in that class. Will do it tonight.

profezzorn, 2024-06-12:
I'm confused. ServoSelector is checked in....
And ProffieOS seems to compile.
Did you do a partial sync or something?

NoSloppy, 2024-06-12:
Not at all. I just fetched again, same thing:
```cpp
In file included from /Users/bconner/Documents/GitHub/ProffieOS/ProffieOS.ino:617:
/Users/bconner/Documents/GitHub/ProffieOS/blades/simple_blade.h:316:31: error: 'ServoSelector' does not name a type
  316 | template<int pin, class LED = ServoSelector>
      |                               ^~~~~~~~~~~~~
```
It's got all the commits

profezzorn, 2024-06-12:
Ah, I see it now. I forgot I changed the include order...
Should be fixed now.

closed 2024-06-12 by profezzorn