profezzorn/ProffieOS #666: new color change mode doesn't like ColorChange<>

NoSloppy, 2024-06-25:
In the past, clicking power when there was ColorChange<> in the blade style would choose the next color in the list and exit.
That's not included in the new mode.
However that's not what I'm actually reporting.
In trying to just rotate through the colors in the list instead, I found it will change once or twice, then start to cycle uncontrollably through them.
This is the blade style in the following video:
`ColorChange<TrInstant,Red,Green,Blue>`

https://github.com/profezzorn/ProffieOS/assets/53964195/99f5f442-5d52-4699-86f7-240cb2e3f921

profezzorn, 2024-07-01:
So, I haven't had time to test it, but I took a look at the code and found lots of bugs.
I re-wrote the buggy part, which hopefully means that at least the bugs are different now.... :)
Anyways, if you test it, let me know how it works, otherwise I'll probably work on it more in a few days.

profezzorn, 2024-07-03:
I did some basic testing, and this seems to work as far as I can tell.
Closing this issue.

profezzorn, 2024-07-03:
actually closing...

closed 2024-07-03 by profezzorn

NoSloppy, 2024-07-03:
Working really well too it seems! 
ColorChange fixed, and smooth is now less sensitive, yeah? Takes more rotation, therefore easier to target a color, similar to "zoom".
Great.

profezzorn, 2024-07-03:
Yeah, the idea was that with three rotations, zoom might not be needed, making the controls simpler.