profezzorn/ProffieOS #749: `sd 1` when ON still allows idle.bmp image to show

NoSloppy, 2025-03-03:
A second issuance of the `sd 1` command is needed to stop idle.bmp and allow the SD to mount.

profezzorn, 2025-03-03:
How about now?

NoSloppy, 2025-03-03:
> How about now?

not yet. same behavior.

profezzorn, 2025-03-03:
How about now?

NoSloppy, 2025-03-03:
> How about now?

Yes. SD access works. idle sound and image both stop if `sd 1` is issued whether ON or OFF.
However, "sd access" is only shown if it was ON when sending the command.
OLED just goes blank if was OFF. Minor, but do we want it consistent?

profezzorn, 2025-03-03:
It's because "sd 1" just triggers the same behavior as IDLE_OFF_TIME, so the oled doesn't really know *why* it was asked to turn off.... Consistency would be nice, but I'm not sure it's worth the extra complexity.

closed 2025-03-03 by NoSloppy